spegelius/filaswitch #1: Initial Testing

ericsammut, 2017-07-06:
The initial test had some issues:

1) There was an unusual delayed response from when the print file was selected until the printer actually started the print. (Not a serious issue, just odd behavior). 

2) Once the print started, the initial movement to the middle of the bed was much slower than the travel speed set in Simplify3D. (Once again, not a serious issue, just odd behavior). 

3) Once the nozzle reached the origin, Extruder 1 started printing the support structure for the purge trail without issues. However, as soon as it finished printing the purge trail supports, Extruder 1 retracted the filament quickly (about 50mm) and then started "trying to print". The motor almost looked like it was skipping steps, even though this was not the case (it was just buzzing and moving back and forth with short movements that seemed to correspond somewhat to the required printing movements that should have been executed).

4) Issue 3) continued until Extruder 1 moved to the purge zone to execute the PURGE_RETRACT movement. This G-code was executed flawlessly, along with the FEED movement for Extruder 2. When the nozzle moved back to the part to print with Extruder 2, the same issue as detailed in 3) occurred again. 

I am not sure if this is a result of the "Relative Extrusion Distances" enabled in the S3D Slicer config. See attached files for debugging purposes. Maybe there is a simple solution that I am missing! :D 

[TEST.zip](https://github.com/spegelius/filaswitch/files/1126563/TEST.zip)

spegelius, 2017-07-06:
1. Hmm is the start g-code the same what you usually use? Just wondering if the hotend and bed heating is done differently...
2. Currently filaswitch uses fixed 150mm/s speed to move to and from purge zone. I'll be adding detection for configured speeds in the future so it should move at proper speed.
3. and 4. Your start g-code:
G90
**M83**
M106 S0
; START SCRIPT START
M104 S260  T0
M140 S[bed0_temperature]
M190 S[bed0_temperature]
M109 S260  T0
G28    ; Home extruder
G90    ; Absolute positioning
**M82**    ; Extruder in absolute mode
G92 E0    ; Reset extruder position
; START SCRIPT END

I've highlighted the problem areas: first S3D sets relative extrusion distances on (M83), but then absolute extrusion distances are set with M82. Remove M82 from start g-code and it'll hopefully fix these problems.

ericsammut, 2017-07-06:
I removed the M82 command from the start script and now issues 3)/4) are solved! The delay in issue 1) is not a problem and is likely a result of the temperature stabilizing before the print starts. 

Issue 2) is still there. I opened the G-code and I found out why the travel speed is so slow. These are the lines of G-code generated just after the start script:

; START SCRIPT END
; TOWER RAFT START
G1 Z0.500 **F2000**    ; z-hop
G1 X-20.249 Y30.746 F9000    ; move to purge zone
G1 Z0.2 F1500    ; move z close

As you can see, the z-hop is executed at 2000 mm/min even though it is executed at 12000mm/min (the travel speed) everywhere else in the G-code. Since I am testing on a delta it means that the end effector lowers to the bed very slowly at the beginning of a print, allowing the nozzle to ooze more filament before the print starts. Is this something that is being changed in the post-processing script?

There is one last issue that is likely just something related to S3D. When I have the printer prepared for a print and connected to the S3D manual control panel, the heat bed turns off as soon as I start the print via SD card. I have to turn the heat bed back on manually through the control panel as soon as I start the print before I remove the USB cable from my computer. 

Besides these minor issues, the post processor seems to work very well!

spegelius, 2017-07-06:
Ah yes, for the raft the z-speed is hard coded. I'll change this along the other travel speeds once I get them read from settings.
I haven't used S3D manual control panel much, but I've noticed that when initiating usb connection it usually resets the controller, at least when connecting the first time. Perhaps this something similar, S3D/firmware wanting to start on clean slate or something...

ericsammut, 2017-07-06:
Another thing I forgot to mention: 

After the filament is switched, the nozzle is primed by printing the purge trail back and forth. This print speed is set a bit too fast as in some cases the plastic did not have enough time to melt resulting in a very thin extrusion width by the end of the purge trail (at the far side of the purge tower). This issue was more apparent with my E3D V6 Upgrade setup which has a hardened steel nozzle with lower thermal conductivity. This resulted is some issues printing small features when the nozzle travels back to the printed part as there is not enough molten plastic in the melt zone to allow for consistent extrusion. In addition, the drastic change in flow rate from the purge flow rate to the printing flow rate results in quality issues like "ringing" on the printed object. This is a known phenomenon with thermoplastic extrusion and is why some slicers have an option for fixed flow rate printing for better quality results. Also, the plastic did not have enough time to heat up in the nozzle which resulted in brittle printed features.

It looks like the purge speed is currently set at 50mm/s or 60mm/s. Reducing the purge speed by about 20mm/s should solve this issue.

spegelius, 2017-07-06:
Yes, purge speed is 50mm/s currently. Didn't know about that, but it does make sense. I'll change it to ~30mm/s.

ericsammut, 2017-07-06:
On second thought, a better solution to the issue I posted above would be to lower the purge speed by only 10mm/s (to 40mm/s) after the filament switch, then for the last purge trail slow the purge speed down to 20mm/s so that the purge flow rate matches a standard printing flow rate. This will allow the nozzle pressure to equalize so that there are no artifacts/ringing when the nozzle returns to print the object, and it will not slow down the purge sequence too much.

ericsammut, 2017-07-06:
Actually, make the last purge trail speed 15mm/s instead of 20mm/s. 15mm/s is what I used for the "drip trail" immediately after the filament switch because it matches printing flow rates most accurately.

spegelius, 2017-07-06:
Ok sounds good. Should the infill-layer have similar speed characteristics?

ericsammut, 2017-07-06:
That would be a good idea. 

Also, these speeds I am recommending are assuming that you are printing the purge trail with 0.2mm layer height and 1mm layer width. With these settings, 20mm/s purge trail speed would result in a flow rate that would match a typical printing flow rate. If you are using the same extrusion width for the infill layer as the purge trail then you can use the same speeds (40mm/s for the outer perimeter of the infill layer and 20mm/s for the inner infill to save time).

spegelius, 2017-07-07:
Yes I've actually been thinking about printing the purge tower at the speed that is configured in S3D as the default printing speed. However I didn't think about the flow rate. But it could be possible to parse the flow rate S3D is using and tune the speeds based on that flow rate. I had something like that in mind when I started this, but then thought that good default values should be good enough.

I'll tune the speeds for now, automatic detection later

spegelius, 2017-07-07:
I released v0.6. Let me know how it works

ericsammut, 2017-07-07:
Testing for v0.6 is going well. Initial z-hop seems to be executed at the proper speed and the heat bed no longer shuts off when starting a print! However, still a few minor tweaks needed:

1) Slowing down the speed of the last purge trail is not enough to fully equalize the nozzle pressure (this is especially apparent with the E3D V6 which has a longer melt zone). As the flow rate slows down, the plastic is melted at a faster rate than it is being extruded so it fills the melt zone in the hot end instead of being pushed out the nozzle. This happens because the melt zone cavity has a 2mm diameter while the filament has a 1.75mm diameter. This is an easy fix, adjust the purge trail G-code so that the last **2** purge trails are printed at the slower speed to allow more time for the pressure to equalize. 

2) Not sure what is going on exactly, but it seems like the wipe feature is not working for retractions executed while printing the single color bottom of the benchy. I am testing with a 2 color benchy model and I do not see the nozzle execute a wipe after printing the outer perimeter for the first part of the print ( I am printing inner to outer perimeters). I have a 4mm wipe configured. The wipe seems to work on the dual color portion of the print because I am not getting any stringing... I am not sure exactly when the wipe is working and when it isn't... it is very hard to tell because I have very short 200mm bowden tubes so I don't really get stringing, even without using the wipe feature.

spegelius, 2017-07-07:
Great to hear, dunno how the heatbed was fixed tho :D.

1. Ok I'll slow two last purge lines to 15mm/s. Small fix
2. Hmm you'll have to contact S3D about when the wipes are executed, I'm only fixing the wipes already there :).

ericsammut, 2017-07-07:
Sounds good! Yeah, I wouldn't be surprised if the S3D wipe feature has bugs. S3D seems to have a lot of bugs that aren't fixed...

spegelius, 2017-07-07:
Fix committed. You can download latest master from this link: https://github.com/spegelius/filaswitch/archive/master.zip

Yeah, S3D is buggy. But apparently there's new version on the works judging by their twitter: https://twitter.com/simplify3d. Let's see how it turns out

ericsammut, 2017-07-07:
Ok, it looks like more tweaking is needed for the slow purge! Make the slow purge speed 20mm/s instead of 15mm/s and make the last **4** purge trails run at the slow speed of 20mm/s. This should give enough of a buffer to eliminate any pressure differences. This should be good enough for a short term solution until you can find a way to autocalculate the purge speed to maintain constant pressure in the nozzle.

spegelius, 2017-07-07:
Went a little bit different route; as I mentioned in forum post, I wanted to print a wall around the purge lines to keep things in shape so I added it now. It means a bit more purge material. So now there's like 4 purge lines printed slow in total and also the previous lines are printed so that the speed is gradually slowed from 2400 to 900 (40mm/s to 15mm/s). I haven't tried it yet, but the g-code looks ok and S3D preview seems ok too. I'll test it tomorrow, can't start a print today anymore.

ericsammut, 2017-07-07:
I like your solution better. I did some testing and here is a list of minor issues/solutions (we're getting there!):

**1)** **For E3D V6 Code only**: There is still not enough time to restore pressure in the nozzle (the back side of the purge tower perimeter does not print because the pressure is still equalizing in the nozzle as the filament melts). In addition, the last purge trail ends at the back right corner of the purge tower, but the nozzle has to move to the back left corner to start printing the perimeter resulting in an unnecessary travel move.

->Solution: Add an extra purge trail so that the nozzle ends at the back left corner (this will add 1mm to the tower width but that's ok). In addition, adjust the speed gradient so that the minimum speed is maintained for the last **5** purge trails (it is currently maintained for the last **3** purge trails). This will give the E3D hot end the extra time it needs to reach equilibrium pressure in the nozzle. 

**2)** 15mm/s minimum purge speed is too slow because I am getting ringing on my prints which is caused by the increased feed rate when the nozzle starts printing the object. This is dependent on print settings so it will vary for each user. I am currently printing with 0.5mm extrusion width, 0.2mm layer height, and 60mm/s print speed. Assuming you are printing the the purge trails with a 1mm layer width and 0.2mm layer height, 30mm/s purge speed would actually be optimal to maintain constant pressure. However, for someone printing at 30mm/s print speed, 15mm/s purge speed would be optimal to avoid ringing. 

->Solution: For now just pick a fixed purge speed that is most appropriate for standard flow rates (15mm/s may be a bit slow). The autocalculated minimum purge speed will solve this issue later.

**3)** The purge tower is slightly offset to the left on the raft printed below it. 

->Solution: You may want to center the purge tower on the raft to improve adhesion if it's not too much trouble. 

**4)** Only one perimeter is printed for the purge tower raft. I find that it takes more than this single perimeter to prime to nozzle. Additionally, sometimes the raft does not adhere well enough to the bed because the raft lines are not held together.

->Solution: Add 2 perimeters to make a mini brim for the raft which will help prime the nozzle and improve adhesion.
- 

Besides these minor fixes, the post processor is coming along very well. The new enclosed purge tower looks super cool :D

spegelius, 2017-07-08:
Committed some fixes.
1. Done
2. Set min speed to 20mm/s. Probably needs more tuning for the gradient, didn't touch it yet. Also fixed probable bugs with extrusion, had - sign on one place so it's possible that there was retraction on the last line instead of extrusion (E3DV6)
3. Centered
4. Added another perimeter. I have PEI sheet on my bed and the raft sticks quite well. However, it might help if the feed rate is increased 10-20% for the raft if needed.

ericsammut, 2017-07-09:
Great! I did some more testing and these fixes really helped. At this point, the purge tower works really well, and the prints look good. However, there is some "blobbing" at points in the print, but I would not be surprised if this is a result of bugs in S3D... 

Also, the minimum purge speed seems to have a huge effect on print quality. 20mm/s improved the print quality, but I still notice overextrusion on the features the are printed immediately after the purge tower. I think the autocalculated purge feed rate is the only major task left to perfect your post processor. No rush with this! 

I think there was indeed a negative feed rate in the old E3D V6 specific code, because your new fixed version does not have any issues with stalled extrusion. I noticed that the last 8 lines are printed at minimum purge speed for the E3D V6, and the last 7 lines for the Pro V2. You could definitely get away with only printing 4 and 3 lines at min speed, for the V6 and V2, respectively. That may save some print time and the 4:3 ratio also matches the relative melt-zone lengths of the V6:V2 which makes sense.

The extra perimeter really helped the raft. No need for increased feed rate.

I have one small request with regards to the GUI just to improve ease of use with our User Guide. It would be great if you could change the HW configs on your GUI to match the config names used in User Guide:

PTFE-PRO-12
PTFE-EV6
PEEK-PRO-12 (Old Config)

Thanks!

spegelius, 2017-07-09:
Committed couple of fixes:
- purge min speed lines tuned to 3 and 4 with E3D
- post tower prime reduced by 0.05 mm for testing. Should help with the over-extrusion
- changed the GUI element names

spegelius, 2017-07-12:
I've made lot's of changes to purge lines. Also added the automatic detection of feed rates. See latest release. If it seems good, I'll close this ticket

ericsammut, 2017-07-12:
Sounds great! 

I just tried testing and for some reason, the post-processor will not work with my g-code files. I am getting an error: **ERROR - division by zero** This error is preventing the post-processor from opening the file.

I am not sure what caused this. Maybe one of my S3D settings is incorrect? Let me know what you think.

spegelius, 2017-07-12:
Ok, send me the g-code file and I'll dig out the error. It's probably in the feed rate calculations, just need to add some checks

ericsammut, 2017-07-12:
Ok, here it is:
[ERROR.zip](https://github.com/spegelius/filaswitch/files/1143386/ERROR.zip)

spegelius, 2017-07-12:
fix committed to master

spegelius, 2017-07-13:
master had stupid bug, caused one purge line to have slightly too much extrusion (about 100000 times too much :D). Shouldn't code late at night... oh well, it's fixed now

spegelius, 2017-07-13:
v0.8 released, should be working now.

spegelius, 2017-07-17:
If these problems are now fixed, I'll close this one. For new errors, please open new issue per problem, easier to track them that way.

ericsammut, 2017-07-17:
Yup, it looks like all the issues are fixed. You can close this one!

closed 2017-07-17 by spegelius