spegelius/filaswitch #5: Support other than 0.2mm layers?

spegelius, 2017-07-17:
Higher layers are possible, lower than 0.2 mm might cause problems with the pre-switch purge (too much material to deposit on 0.1mm layer). Wider paths maybe, but probably will cause unwanted blobbing

gavatron3000, 2017-11-28:
So the pre purge must be completed at a particular rate? I mean the 0.1 would be excellent but can't have the cake and eat it too. I'd be happy to "risk" it but  if it could be done over larger area and purge the right amount of material do you think it would be achievable or is that the limiting factor?

spegelius, 2017-11-28:
It's not the rate, at least it isn't the limiting factor. It's the amount of material to purge. With 0.1mm layers it means double area for the purge tower and it's doable, I'll just have to calculate the tower size differently. Also the initial fast purge might be bit problematic because it needs to be done at fixed speed/rate, need to figure something out...

gavatron3000, 2017-11-28:
thought the fixed speed/rate to be the limiting part. maybe widing the tower and including a one extrusion gap between each printed line and offsetting each following layer by one line as to "simulate" a 0.2mm layer but still incrementing 0.1mm .. just brainstorming out loud

spegelius, 2018-04-10:
Fixed in 0.18.0

closed 2018-04-10 by spegelius