spegelius/filaswitch #9: Odd Purge Tower Preview...

ericsammut, 2017-07-26:
I am still using the old S3D and Filaswitch v0.8, but I have noticed something odd in the G-code preview. 

As you see with the Pikachu print preview in the attached image, the purge tower does not correspond to the filament switches required for the print... The purge tower is also much higher than necessary. I am not sure what is going on here.

[image]

[Pika Files.zip](https://github.com/spegelius/filaswitch/files/1178292/Pika.Files.zip)

spegelius, 2017-07-27:
I'm fairly sure this happens when S3D dual color print is not configured properly and it does filament change for twice for every layer.

I can reproduce this when I don't use the dual color print wizard and set the processes manually; for example, if I configure either of the processes to use extruder 1, but then there's extruder 2 set to print infill or supports... then S3D must do extra filament switches to first print perimeters with ext1, but then switch to ext 2 for infill.

I always use dual color print wizard, it usually configures the processes properly

ericsammut, 2017-07-27:
Thanks, switching to the dual color print wizard solved the issue, but it's unfortunate that the wizard does not allow for support material generation.

You can close this issue.

spegelius, 2017-07-27:
Great. The support material setup is something I haven't tried yet. I do have some water soluble materials, need to do some testing.
If S3D works always like this when using second extruder for support material, some changes need to be done how the purge tower is built, perhaps expand the tower so that two material switches can be done on one layer

closed 2017-07-27 by spegelius