spegelius/filaswitch #10: Outer perimenter speeds are not detected.

prdatur, 2017-07-31:
You script depends on the comments "; layer" which are filled by S3D.

Print file explaination:
- One coin
- One extruder should print bottom part 0.6mm in z
- Other extruder should print middle part 0.8mm in z
- First extruder should print top part 0.6mm in z

Following situation WORKS:

Color 1, First Extruder, Skirt on that activated.
Color 1 process is set to print the bottom and top parts and skirt, Color 2 to middle part.
Layer 1 will have skirt + inner + outer perimeter

Following situation DOES NOT work:

Color 2, Second extruder, Skirt on that activated
Color 1 is set to print the middle part, Color 2 to print the bottom, top part and skirt 
What now happends is that S3D does a bit wrong in commenting.
Layer 1 uses tool T1 which is correct but only "prints" the skirt and before it also says "processing color 1" instead it should be "Color 2" <- Error from S3D.
Layer 2 usese also T1 which is also correct but only prints the "first layer" of the part.
so Layer 1 has no inner / outer parameter and because real Layer 1 is printed finished with commented "Layer 2". All layers are a bit wrong.

The error occures here:
switch_tower.py Line ~ 388
`yield gcode.gen_direction_move(dir_2, purge_length, min_speed, new_e, feed_rate=feed_rate), b" purge trail"`

From there to gcode.py ~ 254
`yield self.gen_extrusion_move(x, y, speed, e_length)`

There it tries to use the "speed" parameter which is None and therefore not convertable to double.

The speed readout is here: layer.py ~ 122
`def get_outer_perimeter_rates(self):`

Because of the issue of the wrong layer comments, some commented "hole layer" have not included the "outer perimeter" comment.

Possible solutions:
First:
- Store last extrusion speeds within a global variable after ich file load and use this speed if "none" is detected. Pro: Easy to implemented, Con: If so much layers are bad, the tower could go very bad.

Second:
- Do not rely on entirely on "layer comment", use the layer comment as a "layer marker" and parse the z value, if it is the same then last one, append all codes to the previous layer and do not create a new layer.
Pro: Mostly it will really "heal" the layers and will not damage anything, Con: A bit complicated

To better see the Problem I have attached the two sliced files.
CoinsTest_CBlack.txt: NOT working
CoinsTest_CBlack2.txt: Working

[CoinsTest_CBlack.txt](https://github.com/spegelius/filaswitch/files/1187745/CoinsTest_CBlack.txt)
[CoinsTest_CBlack2.txt](https://github.com/spegelius/filaswitch/files/1187746/CoinsTest_CBlack2.txt)

spegelius, 2017-07-31:
Ah yes, I actually hit similar problem previously but fixed it in some qnd way and forgot about it. Yes, the initial layers and S3D's empty layers need more attention. I'll probably do a second pass and set all null values to the speed of previous non-null speed value. Also need to check if first layer needs some special handling...

spegelius, 2017-07-31:
Fix committed to master

spegelius, 2017-08-08:
Released in v0.9

closed 2017-08-08 by spegelius