spegelius/filaswitch #11: Strange Z Height behavior

maxstier, 2017-09-08:
Maybe I have things misconfigured or do not understand the system but I'm getting unexpected layer heights for the first layer of the print.  The order of events are that the prime tower prints two layers then the model begins to print.  At this point the nozzle is already up around 0.4mm so my prints aren't sticking to the bed at all.  The prime tower looks great and those lines are a great height off the bed, but yeah the models themselves begin printing at layer 2 and thus off the build surface.  I can't think of what I'm doing wrong. Simplify3D v4 (Mac) running python3.6 for your scripts.

spegelius, 2017-09-08:
Hmm seems strange. Do you have z-hop (retraction lift) enabled? I haven't tried filaswitch without z-hop so this might be the reason. If possible, send me the original gcode file so I can debug this.

maxstier, 2017-09-08:
I don't have z-hop enabled.  Here is a Onedrive link with my gcode ( https://1drv.ms/u/s!Ahc-zrgz6-gwzSRLtCYlZEAxN4vB ).  I really don't know enough to do much troubleshooting myself.  It does look like two layers of the prime tower are being laid down prior to the part getting printed, which seems to be why the Z value is 0.4mm rather than 0.2.

maxstier, 2017-09-08:
Hmmm looking at the Gcode myself right now, it looks like the prime tower is printing with a raft, but my part is not.  That would account for the offset of z heights.

maxstier, 2017-09-09:
Update: I ended up finding and loading the E3D V6 .fff from Distech (thingiverse #2067492).  Using that as a starting point with lots of tweaking to fit my machine (its a delta config originally), I seem to be getting good Gcode now with the correct layer heights.  Still not sure what my issue was with my DIY implementation but I am having good results now!

FireBookDuo, 2017-09-12:
[Zhop-Sample-Gcode.zip](https://github.com/spegelius/filaswitch/files/1294775/Zhop-Sample-Gcode.zip)
There is strange Z behavior if "Retraction Vertical Lift" (aka Z-hop) is set to 0 and a raft is added with a separation distance other than 0 in S3D.

I've attached Gcode samples with Z-hop on and off, and when processed, the Gcode with Z-hop set to > 0 is fine, but when is is set = 0, the layers are not spaced evenly.

spegelius, 2017-09-12:
@maxstier, @FireBookDuo: I haven't checked the code yet, but I'm quite certain that when x-hop is either 0 or off, the layer after printing the tower won't work properly. I'll probably need to add a check that adds G1 Z move command to proper height. I'll try to do this this week. For now, the workaround is to use non-zero z-hop value

spegelius, 2017-09-13:
Latest master has fixes that should help with these problems. Please test it and report back

FireBookDuo, 2017-09-14:
Still having the same issue, but it's not as bad with the latest revision.

spegelius, 2017-09-14:
Ok, I'll have do a test print because I can't see the problem just by looking at the g-code or the S3D preview

spegelius, 2017-09-14:
Printing and only problem I can think of is that at the bottom, the 0.5mm raft layers cause the tower to be lower that the actual model, but after a while, the tower catches on. The tower is printed using fixed 0.2mm layers. Is the z-height difference between the model and the tower is a problem?

spegelius, 2017-09-14:
Improved the infill printing, tower z should follow model z more closely now. Not tested, however...

maxstier, 2017-09-14:
My printer is out of commission at the moment (fried my ramps...) but I'll be testing your changes as soon as I'm running again!