spegelius/filaswitch #14: Unknown ERROR reason

goeland86, 2017-10-01:
Here's a problem I'm having trying to run FilaSwitch... The console shows this:
`----------------------

Detected Simplify3D format

HW config: PTFE-EV6

Simplify3D version 4.0.0

Tower start coordinate: X-29.000, Y21.000, position Top

ERROR: 1
`

The S3D factory file is attached, as is the input Gcode I'm passing in. Did I miss something in the settings?

I'm running a custom delta with a 350mm diameter bed and 440mm printable height.

Any input and clarity about what's causing that error is appreciated. I'm no good with Python or I'd debug myself and create a PR eventually.

[filaswitch_error1_bug.zip](https://github.com/spegelius/filaswitch/files/1347251/filaswitch_error1_bug.zip)

spegelius, 2017-10-01:
Hmm interesting printer profile you have; only one configured extruder. I suppose it could work like that, changing the tool head index for the one extruder for different profiles... but currently not supported. I'll have check if this is feasible. If you want to get this working soon, I suggest creating new printer profile with 2 extruders, current one is for 1 only. And it's easiest to use the dual color wizard after that

goeland86, 2017-10-01:
Huh, odd. I could've sworn I set the printer profile to have 2 extruders, or the dual-extrusion wizard wouldn't have worked... But I'll take a crack at it again with a new profile.

spegelius, 2017-10-01:
Well S3D can be bit obscure about when a printer is configured to have to extruders... for example, you cannot modify existing profile to 2 extruder profile, you have to create new and select 2 extruders. Or modify xml files, but that's not very convenient. You do have the option to add extruders to profiles, but in my experience it doesn't work as well as having a real 2-extruder profile

goeland86, 2017-10-01:
Right you were, adding another extruder fixed the problem!

Thanks!

closed 2017-10-01 by goeland86