spegelius/filaswitch #16: New tower postition introduced in v0.11+ breaks first print layer on Delta

vonkoda, 2017-10-08:
After testing Filaswitch Gcode on every version since 0.08 when I knew it worked I can confirm the slamming into the print bed bug was introduced in v0.11, I suspect with the new tower placement logic (as its way further away). I also note its frequently not 5mm away any longer, on my Rostock Max V2. 

Its almost like it not longer respects the Horizontal Radius value on delta's and tries to print in a weird concave shape. Its unusable on Delta's now!

spegelius, 2017-10-09:
I'll have a look. If you have g-code with this behaviour, please send it to me

spegelius, 2017-10-09:
Not sure if related but I found a small bug that manifests when z-hop not enabled. Fix pushed to master.
BTW, do you have some kind of z-offset defined in printer profile? Filaswitch doesn't recognize that yet, but I'll add support for it

vonkoda, 2017-10-09:
Z-hop meaning lift on retract? I dont use it indeed, maybe related. 
Also Z-offset, I tried to set it to help and sure enough Filaswitch seemed to ignore it! 

Attached a bunch of Gcodes, same model and location, everything past 0.10 hits the build plate. Im changing the build plate tomorrow as this one has taken a bit too much wear and tear (its a PEI plate)!

[pikachu dual.zip](https://github.com/spegelius/filaswitch/files/1369712/pikachu.dual.zip)

spegelius, 2017-10-10:
Do you have the original, non-fs file? Would like to run filaswitch and debug.

spegelius, 2017-10-10:
You seem to be using 0.18mm layer height. I'll add check for 0,2mm layer, fs will fail from now on if layer height is something else

vonkoda, 2017-10-10:
Original file attached if you still need it! 

So no printing tighter layers? Any reasons why? I was trying to push quality ever so slightly higher, or at least see how good I could get things! 
[dual.zip](https://github.com/spegelius/filaswitch/files/1373529/dual.zip)

vonkoda, 2017-10-10:
[pikachu dual.zip](https://github.com/spegelius/filaswitch/files/1373532/pikachu.dual.zip)

spegelius, 2017-10-11:
Tower is printed using fixed 0.2mm layers and if using smaller layers for the actual models, the tower will grow taller and might cause problems with some printers (like mine): the part cooling fan or other parts will start hitting the tower. 
There is issue open about supporting other than 0.2mm layers that I made some time ago as I'm planning on supporting it, but that will need work as I need to widen the tower when going to sub-0.2mm layers (more material to cram on smaller layer). Going to higher than 0.2mm isn't as big of an issue, but not tested and might cause problems with tower walls not sticking properly.

spegelius, 2017-10-12:
Ok, I found one bug related to the tower positioning and it wasn't delta specific; I was using X axis max coordinate also for Y axis max, which moved the tower too far when the object wasn't in center.

I also tuned raft printing, it now matches the first layer height so tower and model should match heights during the print.

But I cannot find a reason why the nozzle would hit the bed. I checked all of the files and every one of them has every Z-coordinate matching their layers, i.e. minimum z-coordinate is at least the first layer height. So filaswitch isn't adding a move that would do that. However, there's always possibility that some command is confusing the firmware and printer is doing funky stuff because of that. Does it hit the bed right at the beginning of the print?

vonkoda, 2017-10-14:
Yes,  right at the beginning, about 3mm into the bed too, not just a little.  Have you got a later version (than 0.12) to test out?

spegelius, 2017-10-14:
Download the latest master branch: https://github.com/spegelius/filaswitch/archive/master.zip

Strange, there isn't any G1 Z command that would go below zero... only thing I can think of is that the homing is somehow messed up and thinks that the bed is lower than it really is...

vonkoda, 2017-10-14:
Did a quick test, the tower position is good now!  However it was still hitting the bed.. so I checked out the bed and found the under my PEI sheet the glass was cracked in two places and the glue on the PEI ravelled up!  I think what was happening was the tower placement was moving to an area with a diff level than the test and generally some weirdness with the heat, PEI, cracked glass bed, glue etc.  So many variables! 

Looks like Im going to have to fix the bed first. :(

vonkoda, 2017-10-14:
Update on the previous.  Took out the glass, re-calibrated the Z height, straight to PEI now. Calibration looks good. 

Test print with Filaswitch.. slams in to PEI sheet again. Stop print, lift head, set to new Z height to get page under it.. 2.5mm higher. Re-do exact print.. tower base layer kinda prints ok but once it moves to print outline of part its moving in bowl shape and 2.5mm in the air!

Its like just for the moment of tower (base) printing its not respecting the delta radius!?

spegelius, 2017-10-15:
Do you have the S3D Update firmware configuration-option enabled? Do you print using USB? I wonder if that could mess with printer settings...
I have no experience with deltas so don't know what could cause such behaviour. Been thinking about getting one, though, maybe debugging this would be a good excuse... :)

vonkoda, 2017-10-15:
I try not to have Gcode overrides per Profile and set them in the Machine Options setting instead. Is there a reason I should enable them? 

Last night completed a bit 2 colour print however, with the Build volume set to 260 / Origin Offset 98 (which are not the settings to use according to S3D's printer setup wizard for Rostock)!
[image]
[image]

spegelius, 2017-10-16:
I don't know about the overrides, was just wondering that could cause the problems... I don't print with S3D USB, I save the g-code and print it with OctoPrint.

Print looks good, did the build volume settings fix the problem?

vonkoda, 2017-10-19:
Ok... still having this problem and I think I found it.

Your script has these two lines in the first part of Tower Raft Start:

G1 X0.038 Y36.981 F12000; move to raft zone
G1 Z0.7 F1200; move z close

For reasons I cannot understand, _"G1 X0.038 Y36.981 F12000; move to raft zone"_ is moving in the Z-Axis on my Delta as well as in X and Y if its sent as the first command. 

I tested this by:
- manually sending the GCodes. If I send it to Z Close (G1 Z0.7 F1200), first, then its all good 0.7mm above the bed. 
- if i then send  move to raft zone (G1 X0.038 Y36.981 F12000) afterwards, its fine, it moves to the correct position. 

But as per the generated GCode, sending it to the raft position before dropping down to the bed, messes it all up completely!  Can you switch these positions in your code? I can't see why its doing it, but saying that all other operations on the printer Ive ever done always, exclusively, drop down the Z-Axis first, then position in X and Y, not the other way around.

spegelius, 2017-10-20:
Apparently when the print starts, the print head is on the topmost position? If so, sideways movement will cause the head to go down as some of the actuators cannot move upwards anymore.

No, I can't switch the positions, that's the way they need to be. But I could add a check before moving to raft position with Deltas that if the head isn't moved to near the bed, it's done before moving sideways. Or you can do it in the start g-code, do G1 Z10 F1200 or something to get the head away from the top position

vonkoda, 2017-10-20:
Hmm, that's interesting. How will this work when the start of every print is a G28, resetting the position on all 3 towers? Otherwise it has no idea how high it is off the bed!?

spegelius, 2017-10-20:
Do G28 first, I didn't mean removing it. Then G1 Z10 F1200, which should move the head in Z axis only to 10mm off the bed. The 10mm is just something I think would be good arbitrary point that's isn't the zmax, change it it you feel like it. Filaswitch will move the head to proper position anyways when starting to print the raft

vonkoda, 2017-10-20:
Yeah but I dont want to have to edit the gcode manually each time (as G28 is in the GCode), cant that be added to Filaswitch , which is pretty similar to the suggestion of moving Z before raft position XY?  Maybe Im not following your suggestion? ;)

spegelius, 2017-10-20:
I meant add that G1 Z10 command to the S3D start g-code section, the G28 should be there already. But I'll add it to filaswitch, it shouldn't hurt to do it on all printers.
I run G28 and then G29 on my printer, which moves the head close to bed during the autoleveling so head is close to the bed right from the start. But my printer is cartesian which doesn't have this problem anyways as the print volume is cube, opposed to delta's cone

vonkoda, 2017-10-20:
Ah yes, that makes sense as long as there are no other G28's floating around. Ill give it a shot later, thanks!

spegelius, 2017-12-02:
Added z-move before printing raft, should fix this

closed 2017-12-02 by spegelius