spegelius/filaswitch #17: Suggestions

cfelicio, 2017-11-19:
Hi There,

First of all, thanks so much for developing this! I have been using S3D, and got a bit tired of the default tower, as so much material goes to waste!

I have a few ideas to improve the script. For now, I modified those directly on the code, but if you could make them in variables (and put on a configuration file), or allow them to be tweaked on the GUI, that would be awesome! :-)

1 - under switch_tower.py, starting at line 545, you have the speed of the raft wall hard coded at 2000, and the rafts at 1000. Some people have finicky bed adhesion, so if this can be changed to either match the first layer speed (defined on S3D), or to be configured, it would improve the raft bed adhesion, for those who need to print it slower. Also, a bit more tricky, but some people like to increase the line width for the first layer, so it's also something that could be changed here

2 - starting at line 392 (and repeating for multiple lines), you set the purge feed rate at 6000. For some people, that's a bit too fast, and it would be cool if it could be tweaked

3 - The code for line 401 to 404 is also something that would be helpful to be able to tweak, as for example, for myself, I found that increasing the retraction speed to the following works much better:

            yield gcode.gen_extruder_move(-55, 8000), b" rapid retract"
            yield b"G4 P2500", b" 2.5s cooling period"
            yield gcode.gen_extruder_move(-105, 3000), b" 50mm/s long retract"

Cheers and thanks again for such an awesome tool!

cfelicio, 2017-11-21:
Adding to this, it would be cool if the tower can have variable height (following the primary layer height and regular height of the model).

spegelius, 2017-11-21:
1. Following the first layer speeds should be easy to do, extra variables and speed overrides bit more work but nothing major. I'll need to implement a config object to hold all user settings to keep things tidy

2. and 3. For the purge speeds, and purge g-code in general, there's already Issue #13 open, in which all purge g-code or values are to be stored in files, thus editable by users. Don't know when I get time to implement, but it's coming.

Variable tower layers: Issue #5 :)

spegelius, 2017-12-02:
Made raft use first layer speed and added control to GUI for controlling the raft extrusion multiplier. Commits in master

cfelicio, 2017-12-02:
awesome, very much appreciated! :-)

closed 2017-12-02 by cfelicio