spegelius/filaswitch #18: Suggestion - Printing Order

cfelicio, 2017-11-19:
Hi There,

Another idea, this one is more on the "optimization" category. The way the script works, it first prints the raft, then moves on to the object you are trying to print, then back to the raft (to switch color), then to the object.

I would suggest to modify the order to start with the object, and then the raft. Why? Because for some users who do not use Z Hop, it is not optimal to print the raft first, as we end up having to hop anyways. It goes like this:

1 - Print the raft at 0.3mm (blue)
2 - Print object at 0.3mm (blue)
3 - Print raft at 0.6mm (switch color to red)
4 - Hop back to 0.3mm to print the red portion of the object
5 - Print object at 0.6mm (red)

If we change the sequence to start with the object:

1 - Print the object at 0.3mm (blue)
2 - Print the raft at 0.3mm (switch color to red)
3 - Print the object at 0.3mm (red)
4 - Print Raft at 0.6mm (switch to blue)

Does this make sense? Or am I missing something?

Thanks! :-)

spegelius, 2017-11-21:
Good suggestion, I've thought about this a while back but forgot to make an issue about it. Should be easy to implement

spegelius, 2017-12-03:
Pushed a commit to master that should print raft right before first actual tower infill or purge layer. Not tested though so not 100% sure if it works

cfelicio, 2017-12-04:
Awesome, let me give it a try and get back to you!

cfelicio, 2017-12-17:
umm, looks like it's still z-hoping. Besides moving the raft right before the first actual infill / purge, it is also necesary to remove the first infill / purge, and print the part first. Not sure if it's easy to do it? I will give it a try, as everything else is working great for me :-)

spegelius, 2017-12-17:
You mean remove the initial raft-layer altogether? The inital raft-layer is printed to make sure the whole tower stick to the bed properly. I'm not sure that the infill-layer or tool switch layer will stick the the bed.
Sure I could make it optional, but it's not a priority right now. Also with S3D, there's probably going to be situations where the model and tower aren't at same height -> z-hopping.

cfelicio, 2017-12-17:
Oh no, the raft stays, but the hopping is only partially addressed. Let me elaborate.

If I print a part that has no dual color on the first few layers, the code will work as expected:

- print part
- print tower raft
- print tower infill
- print part

And so on. No Z hop happens, it's perfect!

Now, when a color switch finally happens (let's say, 30 layers later), the code starts hopping again:

- print part
- hop up
- print tower purge
- hop back down
- print part

So ideally it should go:

- print part
- print tower purge
- print part

Also, in order for this to work with 0 hopping, if there is a color switch on the first layer, the color switch would need to happen before the raft is printed, which means the raft would act as the purge (which I'm not sure if it's ideal, but I'm willing to give it a try if it's simple for you to change).

Hope this makes sense! If you need more info, let me know! :-)

spegelius, 2018-02-12:
Latest master doesn't print raft if tool change is at first layer. There's an option to force it in Advanced settings if someone wants it for some reason, however.
I use z-hop always, less error prone to stick to tower/model and fail so I don't know if z-hopless code works sensibly.

cfelicio, 2018-04-10:
I ended up adapting my retractions to work well with Z hop, so we can close this. Thanks! :-)

closed 2018-04-10 by cfelicio