spegelius/filaswitch #21: stability of tower above 100mm

gavatron3000, 2017-12-05:
so......doing a large print i had the unfortunate event of the tower toppling over during a 60 hr print and tower tipped at roughly 55hrs. tower was 110mm high when it let go
could there be an option to turn the tower 90 degrees to suit my moving y-axis bed better? (standard reprap style). 
regards
gav

spegelius, 2017-12-05:
That's unfortunate, I know the feeling when long print fails at the end...
There are few options already:
- in advanced tab, select tower position Left or Right. Both rotate the tower 90 degrees
- latest code in master branch has option to tweak the raft extrusion amount. Bumping the extrusion to 105% or 110% should help with the adhesion. I'll also make the raft brim size to follow the slicer brim size in future.

Tower rotation option is possible if really needed

gavatron3000, 2017-12-05:
I guess my scenario was unique as the part was large and took up the bulk of the middle of the bed and edge to edge left to right. The script would error when selecting left or right due to the size of the print. I'll try moving the part backwards to see if it will go left or right but at the front corners

spegelius, 2017-12-05:
OK, in that case left/right positioning won't work. The positioning logic doesn't understand empty spaces well enough, it just checks min and max coordinates in x and y directions. Might work in future, but will require whole new level of spatial awareness in the script...
Can you rotate the model 90 degrees or is it too big for that?