spegelius/filaswitch #22: Extraneous extrusion from tower

paukstelis, 2017-12-14:
Thank you for this great software. I'm making some tweaks to use it with the Prusa quad MMU. One thing I keep noticing are extra extrusions that are occurring between switch tower and supports (using S3D to slice). 
[image]

It seems in this case that it begins the un-retract before moving to position. This is the apparent offending gcode:
```
; TOWER INFILL END
; layer 10, Z = 0.800
G1 Z0.800 F1200
G1 E4.0000 F3000
G1 X101.500 Y71.659 E0.1549 F240
```
From the same file but works fine (unretract after moving to position):

```
; TOWER INFILL END
; layer 13, Z = 1.000
; inner perimeter
M204 S1200
G1 X127.646 Y44.118 F9000
G1 Z1.000 F1200
G1 E4.0000 F3000
```

I don't think this is something from my post-processing, though I can't be certain.

spegelius, 2017-12-14:
Hmm, I haven't seen that kind of behaviour before. To me it doesn't make sense to prime the nozzle before moving to next position. Could you send me the original gcode file so I can debug?

Since you are testing PrusaMM with filaswitch, I suggest you try the code from cfgfiles-branch: https://github.com/spegelius/filaswitch/tree/cfgfiles
It moves the purge settings to config files so adding PrusaMM config should be much easier, no need to edit the code. Feedback about the feature would be great and if you get a working Prusa config done, I could add it to the SW if it's ok with you.

paukstelis, 2017-12-14:
Thanks.
Here is a link to the [file.](http://vespa.chem.umd.edu/~paul/e21_ext.gcode). 

I did checkout the cfgfile branch just yesterday, but ended up with errors when trying to run the gui from it and haven't looked into it any deeper. This is from a clean clone.
```
 list index out of range
Traceback (most recent call last):
  File "/home/paul/extra/filaswitch/utils.py", line 28, in load_status
    status[vals[0]] = vals[1].strip()

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "filaswitch.py", line 38, in <module>
    settings = Settings()
  File "/home/paul/extra/filaswitch/settings.py", line 31, in __init__
    self.read_hw_configs()
  File "/home/paul/extra/filaswitch/settings.py", line 160, in read_hw_configs
    self.parse_hw_cfg(cfg_name, path)
  File "/home/paul/extra/filaswitch/settings.py", line 166, in parse_hw_cfg
    cfg = utils.load_status(cfg_path)
  File "/home/paul/extra/filaswitch/utils.py", line 33, in load_status
    raise IOError("Cannot open file %s" %status_file)
OSError: Cannot open file /home/paul/extra/filaswitch/hw_configurations/PEEK-PRO-24.hwcfg

```

spegelius, 2017-12-15:
Ah I think I know what's the problem. Before making the last commit to cfgfiles branch, I added comments to the .hwcfg files. And did the mistake of not testing the code. I just assumed that the comment lines would ignored, but since the files are parsed by my own code, I haven't even implemented  the comment support yet... :). Oh well. I'll fix that during the weekend.

paukstelis, 2017-12-15:
For cfgfiles:

I forked and started making some modifications in cfgfiles. I realized the comments were an issue so just removed them for now. I also had to convert the dict_keys to a list in order for them to be read into the GUI. Made a few other modifications to try and more closely mimic what happens in Prusa-Slic3r gcode for MMU.

For extra extrusion issue:

It looks like this comes from how layer changes occur in the starting gcode, and how it determines where to put the tower infill. I'm testing a work-around in gcode_file that basically reads ahead to see if the next step after the layer change is just a z move. If it is, it will skip putting the tower infill in that spot. Previewing the gcode it seems like it will work OK, but I haven't tested it yet.

spegelius, 2017-12-15:
Ok great, if you get something working, create a pull request. I'll have a look also, seems that cfgfiles need some work. I'll fix the comments so they won't cause problems anymore.

spegelius, 2017-12-15:
I fixed the comments crash and pushed it to cfgfiles. Also had a look at the gcode and apparently some setting that's different than I use on my profiles is causing this prime before move into position. Usually it is ok behaviour, but now that the nozzle is actually somewhere in the tower, it becomes a problem. My solution would be to just move the prime to happen after the head has moved in position, but I'm interested to see how your solution works.

paukstelis, 2017-12-15:
My work-around is pretty ugly, but it seems to work. See: https://github.com/paukstelis/filaswitch/commit/1e13cc79b72b6e17d2b68776bf535bac52872349

Since I've been put cfgfiles branch issues here, I'll also point out that I get different tower raft behavior with cfgfiles vs. master. The extrusion widths of the raft are nearly non-existent:

```
G1 X54.800 E0.0392 F1800; raft wall
G1 Y29.100 E0.0392 F1800; raft wall
G1 X-54.800 E0.0392 F1800; raft wall
G1 Y-28.700 E0.0392 F1800; raft wall
G1 X54.400 E0.0392 F1800; raft wall
G1 Y28.300 E0.0392 F1800; raft wall
G1 X-54.000 E0.0392 F1800; raft wall
G1 Y-27.900 E0.0392 F1800; raft wall
G1 X0.424 Y-0.424 F9000
G1 Y28.300 E0.0510 F1800; raft1
G1 X1.000 F1800; raft2
G1 Y-28.300 E0.0510 F1800; raft3
```
I don't see anything obvious since it should still be calculating extrusion length the same way?

paukstelis, 2017-12-15:
Ahh, turns out that was a fairly easy one. Just needed to change to feed_rate=feed_rate in get_raft_lines of switch_tower :)

spegelius, 2017-12-15:
What is your raft extrusion multiplier in advanced settings? I added that setting fome time ago. Nothing else is changed with raft after that (at least can't remember...)

paukstelis, 2017-12-17:
The raft extrusion issue was a mistake on my part. I wanted to the ability to retract during gen_direction moves so added it to gcode.py and put it in the wrong spot. I didn't realize the syntax for the rafts happened to be different (just passing feedrate) from the pre-change syntax (feedrate=feedrate). 

I'm working on putting the slic3r pre-print priming behavior, then I'll put in a pull request.

spegelius, 2017-12-18:
Ok great. If you have improvements to the gcode-handling functions, let me know. The functions/API whatever can be changed to make it easier/more logical to use.

closed 2018-01-12 by paukstelis