spegelius/filaswitch #23: Issue with nozzle temperature

cfelicio, 2017-12-17:
I'm trying the latest master, and using S3D. For some reason, filaswitch is dropping my nozzle temperature to 60C?

G1 E-55.0000 F8000; rapid retract
M104 S60 T0; change nozzle temp
G4 P5000; 2.5s cooling period
G1 E-105.0000 F3000; 50mm/s long retract
T1; change tool
G1 E10 F1500; 25mm/s feed
G1 E90 F3000; 50mm/s feed
G1 E49 F1500; 25mm/s feed
G1 X-50.000 E4.6500 F900; prime trail
M109 S60 T0; change nozzle temp, wait
G1 E5 F1500; 25mm/s feed
G1 Y0.900 F3000; Y shift
G1 X50.600 E2.1007 F2400; purge trail

Is this something I need to fix on S3D?

spegelius, 2017-12-17:
Are you sure it's the latest master? I just fixed a bug where bed temp was erroneously read as extruder temp.

cfelicio, 2017-12-17:
sorry, you are right, it was a master from a few days ago. I will get the latest one :-)

closed 2017-12-17 by cfelicio