spegelius/filaswitch #24: Weird tower with 3dbenchy with latest master

cfelicio, 2017-12-17:
Not sure what happened :-)

I'm attaching the gcode, plus the processed gcode. Thanks!

[image]

[3dbenchy.zip](https://github.com/spegelius/filaswitch/files/1566183/3dbenchy.zip)

spegelius, 2017-12-18:
Hmm this can happen if the S3D has generated the g-code kind of "wrong". Please send me the gcode so I can verify. I need to update the README about how to set up S3D for dual color printing some day...

cfelicio, 2017-12-18:
I attached the gcode as a zip file, it should contain both the original and the fs files. see above.

Oddly enough, I did not change anything on my process, but I usually move the object from the center, and this time I did not.

I tried to re-generate the gcode with the object off center, and voila, it worked! I did not change anything else besides the location of the benchy! but the gcode generated by S3D seems smaller, very odd!

 
[image]

At this point I consider this solved, but I will leave it open for now in case you want to take a look! Thanks!

[3dbenchymoved.zip](https://github.com/spegelius/filaswitch/files/1566916/3dbenchymoved.zip)

spegelius, 2018-01-03:
I took a look at the files but they seem fine on S3D preview. I did upgrade to S3D 4.0.1 just now, though I don't think that should matter...

cfelicio, 2018-01-03:
umm, I'm still running 4.0. I will try to upgrade tonight. This is a small issue anyways, I will close it for now! Cheers :)

closed 2018-01-03 by cfelicio