spegelius/filaswitch #26: filaswitch - ERROR - 'NoneType' object is not iterable

wbarber69, 2017-12-23:
what is going on here? after the latest windows update i ccan no longer run filaswitch.  this is the only info in the logs

filaswitch - ERROR - 'NoneType' object is not iterable

spegelius, 2017-12-23:
Windows... god I hate all the random problems with it...
But could you run filaswitch.py in cmd? It prints the whole exception that way. I'll need to improve the logs to include that too...
start cmd
cd path\to\filaswitch 
c:\python36-32\python exe filaswitch.py path\to\gcode PEEK-PRO-12
Change python path and hw-config as needed.

wbarber69, 2017-12-23:
I’ll give it a shot but I never had to invoke python before... it’s always worked from the bat file.

wbarber69, 2017-12-23:
C:\filaswitch>"C:\Users\Walter Barber\AppData\Local\Programs\Python\Python36\python.exe" filaswitch.py C:\butter_pig_full.gcode PTFE-EV6
2017-12-23 11:36:11,453 - filaswitch - INFO - Detected Simplify3D format
2017-12-23 11:36:11,453 - filaswitch - INFO - HW config: PTFE-EV6
2017-12-23 11:36:12,534 - filaswitch - INFO - Simplify3D version 4.0.1
2017-12-23 11:36:17,619 - filaswitch - INFO - Tower start coordinate: X124.939, Y222.603, position Top
Traceback (most recent call last):
  File "filaswitch.py", line 291, in <module>
    main()
  File "filaswitch.py", line 286, in main
    result_file = pf.process(args.file)
  File "C:\filaswitch\slicer_simplify3d.py", line 58, in process
    self.add_tool_change_gcode()
  File "C:\filaswitch\gcode_file.py", line 237, in add_tool_change_gcode
    for cmd, comment in self.switch_tower.get_tower_lines(layer, e_pos, active_e, new_e, z_hop):
  File "C:\filaswitch\switch_tower.py", line 747, in get_tower_lines
    new_temp = new_e.get_temperature(layer.num)
  File "C:\filaswitch\extruder.py", line 72, in get_temperature
    for lr in self.temperature_setpoints:
TypeError: 'NoneType' object is not iterable

wbarber69, 2017-12-23:
btw yes i did try this on old gcodes hat i was able to run on before and i get the same issue

wbarber69, 2017-12-23:
and i updated python to latest as well as filaswitch

wbarber69, 2017-12-23:
here is some more possibly useful info.  if i open an older gcode that already has an *fs.gcode counterpart file created for it then the error is posted on screen almost instantly, but if i delete or move the *fs file from the folder and try again, then it takes it about 10 seconds before the errors are posted on screen.

spegelius, 2017-12-23:
Hmm, deleting old fs.gocde doesn't affect how filaswitch works. Maybe something that Windows does... dunno. But the error you posted seems to be related to temperature handling that has been added (and fixed few times) in later releases, Could you send me a gcode-file that fails? That error only tells where it fails, but not why...

wbarber69, 2017-12-23:
It fails on all gcode files, old, new, everything.  Even files with no filament changes

wbarber69, 2017-12-23:
There is a whole new windows update going on right now.  Maybe they realized they broke something.

spegelius, 2017-12-23:
Ok let's see if the updates fix something. Otherwise, send one of the failing files. I'll need one for debugging, there might be something in your S3D profile that causes the crash. Everything is working fine on PC, both Windows and Linux

wbarber69, 2017-12-23:
Nope it’s definitely the latest filaswitch I went back 2 commits and unzipped the older zip file and it ran fine

wbarber69, 2017-12-23:
Could it possibly be something wrong with my slicer?  Maybe it isn’t presenting temperatures right, when I get back home I’ll upload a couple gcode files I’ve made this week...

wbarber69, 2017-12-24:
[butter_pig_full.gcode.txt](https://github.com/spegelius/filaswitch/files/1584599/butter_pig_full.gcode.txt)

spegelius, 2018-01-03:
Finally got time to check this. No errors with latest code from master branch. The error is probably related to a bug I fixed, but haven't released yet. Here's the latest code: https://github.com/spegelius/filaswitch/archive/master.zip

wbarber69, 2018-01-03:
Do you see anything particularly off with the gcode?  For some reason octoprint has been trying to print without waiting for the nozzle temp.  I have it set to wait in my slicer but for the last 10-15 prints I’ve had to manually set my temperature.  But only after I finish or cancel a print, so I don’t know if it’s my gcode or octoprint itself.

wbarber69, 2018-01-03:
I’ll check out the latest code when I get back home, thanks

wbarber69, 2018-01-06:
no dice.

spegelius, 2018-01-07:
Hmm then I don't know why it gives that error. The file you linked processes fine on my PC.
The get the full error, please run filaswitch from command line

wbarber69, 2018-01-08:
C:\filaswitch4\filaswitch-master>"C:\Users\Walter Barber\AppData\Local\Programs\Python\Python36-32\python.exe" filaswitch.py "C:\Users\Walter Barber\Documents\gcode\Printing\CR-10\handle test.gcode" PTFE-EV6
2018-01-07 19:44:15,426 - filaswitch - INFO - Detected Simplify3D format
2018-01-07 19:44:15,427 - filaswitch - INFO - HW config: PTFE-EV6
2018-01-07 19:44:15,778 - filaswitch - INFO - Simplify3D version 4.0.1
2018-01-07 19:44:16,972 - filaswitch - INFO - Tower start coordinate: X125.269, Y235.476, position Top
Traceback (most recent call last):
  File "filaswitch.py", line 291, in <module>
    main()
  File "filaswitch.py", line 286, in main
    result_file = pf.process(args.file)
  File "C:\filaswitch4\filaswitch-master\slicer_simplify3d.py", line 58, in process
    self.add_tool_change_gcode()
  File "C:\filaswitch4\filaswitch-master\gcode_file.py", line 237, in add_tool_change_gcode
    for cmd, comment in self.switch_tower.get_tower_lines(layer, e_pos, active_e, new_e, z_hop):
  File "C:\filaswitch4\filaswitch-master\switch_tower.py", line 743, in get_tower_lines
    new_temp = new_e.get_temperature(layer.num)
  File "C:\filaswitch4\filaswitch-master\extruder.py", line 72, in get_temperature
    for lr in self.temperature_setpoints:
TypeError: 'NoneType' object is not iterable

spegelius, 2018-01-08:
Ok I found the bug. Fixed now. Btw, I added a warning about uncomplete S3D profile, you have temperature only for one extruder. Filaswitch will work but it will nag about having to guess what the second extruder's temperatures should be.

wbarber69, 2018-01-08:
That makes no sense.  The whole point of having a smart prime tower is so you can run 2 extruders through 1 nozzle...

spegelius, 2018-01-09:
No. You still have two extruders which can have different materials loaded, with different temperatures. Whole point of the temperature logic in filaswitch is to support changing nozzle temperature during tool changes, like when printing supports with PVA and main model with ABS.

spegelius, 2018-01-09:
Oh also when you run the printer wizard in S3D and define more than 1 extruder, S3D will automatically generate temperature for each extruder (notice it's not nozzle).

wbarber69, 2018-01-09:
I might need to find a different profile then.  This one was made specifically for a y adapter setup.

spegelius, 2018-01-09:
Not necessarily. Either just let it nag about the missing temperature settings or add it manually. I haven't used the fff profile that's available from Distech site, I created couple of new profiles using the S3D wizard and just tuned the settings manually

wbarber69, 2018-01-09:
I’ll have to look into it, but iirc it was a pain to get s3d to make a custom dual extruder profile without it being a copy of an already dual extruder profile from their site.

wbarber69, 2018-01-10:
Still don’t see how I’m supposed to add a second temperature controller for the same identifier

wbarber69, 2018-01-10:
oh well, that was a failure too.  after i added the second extruder temperature controller now it tries to turn off T0 when T1 is in use.  so now it compiles just fine, but my print fails as soon as the temperature drops out.

spegelius, 2018-01-10:
Hmm, that's weird, I have 4 extruders with 1 nozzle and in S3D, each extruder has temperature settings with identifiers from T0 to T3. Everything works fine. I use Marlin FW with SINGLE_NOZZLE defined so this should make the nozzle respond to all Tx temperature changes. I'll have a look at the printer profile settings closely after work, it shouldn't shut down other extuders when changing filament...
Also if you don't min could you send me the .fff file? You can export it from S3D

wbarber69, 2018-01-10:
yeah i checked my marlin files and i definitely have SINGLENOZZLE defined as well as my number of extruders.  but anything but T0 doesnt change my temp.  im boxing up that fff profile now.

wbarber69, 2018-01-10:
[CR-10 2 in 1 filaswitch.fff.txt](https://github.com/spegelius/filaswitch/files/1619491/CR-10.2.in.1.filaswitch.fff.txt)

wbarber69, 2018-01-10:
I may be zeroing in on the issue.  I think it might be a prblem with marlin 1.1.5 As that was the current branch when [this](https://www.bountysource.com/issues/48214887-issue-with-octoprint-in-a-single-nozzle-heater-dual-bowden-extruder-with-a-y-splitter) eroor post was made.  and i just so happened to stumble across this while researching problems with marlin, and ive been trying to nail down this issue with octoprint as well.

spegelius, 2018-01-10:
Ah you mean that the temp is not shown correctly on Octoprint? That's been a problem with Octoprint and Marlin since I started using 1.1x about year ago. The nozzle still is in correct temp as seen in terminal messages. Of course if it drops to zero on Octoprint it sounds odd, on my setup the it stays on the last T0 temp until switched back to T0.

wbarber69, 2018-01-10:
Right but this doesn’t happen exactly that way for me with the single extruder temperature.  This is why I haven’t found anyone else’s with my issue before.  Because I have unique settings.  I see the temperature failure problem in octoprint but it’s never been an issue till I added the second temperature controller in s3d.  Now it’s functioning as it should but that means that the temperature is turned off when changing controllers.  Which narrows down a lot of this for me.  Now I see that filaswitch won’t work correctly because I’m missing a temperature controller.  But octoprint now is broken because I fixed filaswitch.  So I did some more digging and sure enough my marlin config is set to disable the inactive extruder.  I’ll reflash later and see if that fixes it.

spegelius, 2018-01-10:
Ok, that could be it. I did have to disable that feature too, but I did that only after getting a 4x setup, which uses a relay to switch the extruder (all extruders are actually driven from one stepper driver). My temperatures were working fine even before that. Though I might've upgraded to Marlin 1.1.5 while building the 4x setup...
I checked the fff profile and it seems fine. I sliced a dual color model and there's nothing out of the ordinary so it points strongly to the firmware.

wbarber69, 2018-01-10:
I hope so, because this fff is available to everyone buying Prometheus clones

spegelius, 2018-01-10:
Yeah, the temperature handling is a recent addition so problems might start to surface if people fix their profiles. Though it's not mandatory, flaswitch will still do it's thing even if there's only temperature for one extruder, it uses same temps for all. It just prints an notification about it

wbarber69, 2018-01-10:
yeah thats true, but i want to see this thing working 100% if not for me, then for the hundreds of others who'll soon be pressing this github pretty hard.  Ive been developing a dual color cr-10 and the creality folks have already been in contact with me to help with their soon to be released upgrade kit.  Once i get this nailed down, and design a better extruder mounting system, ill be adding filaswitch to my list of must haves.

spegelius, 2018-01-10:
Interesting, so Creality is also getting to dual color printing. I don't have any Creality printers but nevertheless interested to see the setup. Yes, best to aim for 100%, especially in this case.
I'm running Marlin 1.1.8 currently and just now printing 4 color print, no problems with temperatures. It even seems that Octoprint is showing correct temps all the time. So there's hope yet :)

If you need to tune the retraction lengths, speeds etc. to better suit CR10, check out cfgfiles-branch. It's a big change, it moves the configuration to files that people can modify to suit their needs. I also got some help from a Prusa MM user who added some Prusa type additions and Prusa config. It'll become master branch when it's tested and ready.

wbarber69, 2018-01-10:
That may be an avenue I have to go down some day, but believe it or not the e3dv6 configuration works a treat.  I am using Bowden configuration all the way to the nozzle but the pause for 3 seconds before final pull is an absolute must to keep my y adapter from clogging up.

spegelius, 2018-01-11:
Great. Yeah the sleep is important, otherwise the filament is too molten and will cause problems. PrusaMM has even more functionality, it does head movements during retract and some back and forth filament pulling, apparently to smooth out the head of the filament. Haven't seen need for that with Prometheus, but might be useful for some filaments.

wbarber69, 2018-01-28:
after fixing my firmware and my profile this issue is fixed

closed 2018-01-28 by wbarber69

joaomamede, 2020-01-19:
I have this problem in linux as well

I am trying to use filaswitch with Prusa Slicer. 
I configured everything like in the instructions. 
I have a BTT 2in1 hotend (2 extruders one nozzle). I modified slightly the prometheus Ed6 file for a first try:  
[BTT2in1-PTFE.zip](https://github.com/spegelius/filaswitch/files/4083643/BTT2in1-PTFE.zip)

I then have an error when exporting the g-code:
`Post-processing script /home/myusername/3dprinting/filaswitch/filaswitch/filaswitch.sh BTT2in1-PTFE on file /tmp/3DBenchy_-_Dualprint_-_Gunwale_Deck_Plate_Wheel_Frames_Roof_Chimney_top_-_3DBenchy.com.gcode failed.
Error code: 1
Output:
Traceback (most recent call last):
File "/home/myusername/3dprinting/filaswitch/filaswitch/src/filaswitch.py", line 621, in <module>
main()
File "/home/myusername/3dprinting/filaswitch/filaswitch/src/filaswitch.py", line 604, in main
result_file = pf.process(args.file)
File "/home/myusername/3dprinting/filaswitch/filaswitch/src/slicer_prusa_slic3r.py", line 28, in process
self.parse_gcode()
File "/home/myusername/3dprinting/filaswitch/filaswitch/src/gcode_file.py", line 833, in parse_gcode
self.parse_gcode_pass1()
File "/home/myusername/3dprinting/filaswitch/filaswitch/src/gcode_file.py", line 747, in parse_gcode_pass1
path_len = gcode.calculate_path_length((last_extrusion_x, last_extrusion_y), (current_x, current_y))
File "/home/myusername/3dprinting/filaswitch/filaswitch/src/gcode.py", line 79, in calculate_path_length
x_len = prev_position[0] - new_position[0]
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'`

If I try to do it mannualy with FilaSwitch GUI:
`/home/jmamede/3dprinting/filaswitch/filaswitch/filaswitch.sh BTT2in1-PTFE;`

spegelius, 2020-02-01:
Create a new ticket, this is closed