spegelius/filaswitch #27: lin advance

gavatron3000, 2018-01-06:
has anyone else had any issues running lin advance with the script? extruder stepper has issues during the first retract, using a bondtech qr 1.75. might need to turn down my E acceleration...

gavatron3000, 2018-01-06:
ive worked out i can just use gcode to deactivate before and reactivate after the tool change, so its a non issue

spegelius, 2018-01-06:
I have not heard of such problems. I've only tried lin advance once about 1.5 years ago and it caused problems with even normal printing. Disabled it, but afterwards actually calculated what my settings were doing and noticed that in some cases it was trying to extrude way too much stuff. Haven't tried it since as I see no need for it. But if the feature works like it did, I'd expect it to amplify the fast retracts. You could try the cfgfiles branch and write your own hw config that has lower speeds for the retracts

gavatron3000, 2018-01-06:
No worries if I get a chance I'll have look at that then, cheers

closed 2018-01-06 by gavatron3000