spegelius/filaswitch #28: Odd tower raft behavior

paukstelis, 2018-01-12:
Just notice today that if I place parts below the build plate I got an odd raft dimension for the tower:
[image]

[image]
However, if I sit the object on the build plate, everything is fine.
[image]

Obviously, this isn't a huge issue, but something that might need to be looked at in the future.

spegelius, 2018-01-16:
Hmm strange. I'll have a look

spegelius, 2018-01-16:
I might've found the reason for this; the new preprime messes the calculation for how many max changes are per layer and raft is too big. Fixed it on my cfgfiles branch

paukstelis, 2018-01-26:
Closing this. I tried to reproduce this in the latest cfgfiles branch and it did not happen.

closed 2018-01-26 by paukstelis