spegelius/filaswitch #30: Feature request: Prusa MMU support

laird, 2018-01-24:
The new Multi Material Upgrade from Prusa is only supported by a post-processor for their version of Slic3r. I’d rather use S3D, and from forum posts many others feel the same way.

Is there any interest in adding support for the Prusa MMU to filaswitch?

spegelius, 2018-01-24:
It just so happens that such support is already implemented :). Not on master branch yet, but soon, I'll release new version when the code is tested. Check out cfgfiles branch and the feature pull request: https://github.com/spegelius/filaswitch/pull/25.

laird, 2018-01-24:
I just found https://github.com/spegelius/filaswitch/tree/cfgfiles . Very exciting! My MMU is supposed to show up in a month or two (MK3 MMU isn’t shipping yet, AFAIK), after which I’ll be happy to test, help code, etc.

laird, 2018-01-24:
Since it’s Implemented, should I close this issue? Or wait for the merge?

spegelius, 2018-01-24:
Well technically it's not released yet :). But either way works for me, it'll be released soon anyways

spegelius, 2018-01-28:
Released in 0.16.1

closed 2018-01-28 by spegelius