spegelius/filaswitch #32: cfgfiles: temperature change problems

paukstelis, 2018-01-26:
There seems to be an issue with temperatures changes related to tool/bed with Simplify3D. For whatever reason, every change to T1 issues an M104 to the bed temperature instead of the nozzle temperature.

spegelius, 2018-01-26:
Hmm, check the profile: is bed temp configured as heated bed? I've set bed to be T0 myself and I have 4 extruder temps from T0 to T3.
To my knowledge bed and hot end indexing aren't in same namespace.

spegelius, 2018-01-26:
Actually imported the Prusa fff file, sliced and filaswitched. Yes, there seems to be a bug with temperature parsing

paukstelis, 2018-01-26:
It is not reproducible. I posted my video on the Prusa forum and someone commented that they saw this temp change. I verified that I saw it as well and was trying to get to the root of it. I went back to the beginning, recloning filaswitch and using the profile that I provided there. I didn't see it again. Weird.

spegelius, 2018-01-26:
Btw, the fff file isn't optimal. It must be created from scratch, with the amount of extruders that you have. S3D for some reason saves the profile differently when configured with certain amount of extruders and that can't be changed afterwards (only with editing the xml by hand). This profile will work but it will have more tool changes and more waste

paukstelis, 2018-01-26:
I'll look into doing that. Perhaps that profile is the root of these inconsistencies.

spegelius, 2018-01-26:
There's a bug with the extruder/temp indexing, fixing now. Happens when bed controller is before extruders -> wrong ext
Use the new printer wizard to create new profile. I know it's PITA when you have to copy every setting, been there done that.

paukstelis, 2018-01-26:
Yeah, a bit of a pain :). I'll knock it out, test it,  and submit a pull request in the next day.

spegelius, 2018-01-26:
Great. Pushed a fix for the temperature problem.
Just noticed the support extruder hack, nice :). New profile might change how supports are printed, though so check if it's needed anymore. One thing that the T5 does is make filaswitch nag about missing temperature settings for it, though.

paukstelis, 2018-01-26:
I'm not seeing any differences in the supports with a freshly created profile. I actually think my old profile was created with the wizard, though I added the support extruder after creating it. One thing that is somewhat annoying is that S3D seems to want to cluster the supports with the model that is associated with those supports at the same z-height (i.e. printing order is model1 supports, model1, model2 supports, model2, etc.). It would make a lot more sense and probably speed things up to do model1, model2, model1 supports, model2 supports (or flipped around). I wonder how difficult it would be to do some parsing for all support blocks (they all start with "; support" or "; dense support" and end with the next non-cmd line) and move them to the end (or the beginning) of that layer.

closed 2018-01-26 by paukstelis

spegelius, 2018-01-27:
Ok, I haven't used supports much. Yeah, S3D isn't the optimal multi-extruder slicer and the 4.0.1 version has some bugs which make printing some multi-color models impossible...
I updated the README in cfgfile branch with info about setting up printer profile. I'll add more info about how I set up the processes for multi-material prints later.
About reordering stuff: yesterday new issue was posted about using model infill as purge target: https://github.com/spegelius/filaswitch/issues/33. Got me thinking about the implementation and basically parsing all the different print areas would be needed to allow easier rearranging. Supports would be easier to move around also. Needs some refactoring, but doable.