spegelius/filaswitch #33: Infill Transitioning

cfelicio, 2018-01-26:
Not sure if even remotely possible to implement something like this, but quite amazing feature:

https://www.reddit.com/r/3Dprinting/comments/7t5zx8/mosaic_introduces_infill_transitioning/

spegelius, 2018-01-26:
Hmm, interesting. Yes it's possible but requires some work. Will require more thorough analysis of the gcode and some calculation about infill length and if additional purge is needed...

wbarber69, 2018-01-28:
this can only work if your part has enough infill to do the transition

spegelius, 2018-01-28:
That's true. The implementation would still need purge tower as backup location for cases when there's not enough infill for the purge.
Other possibility that I've thought about is using another model as purge target. The model would have some requirements for optimal usage and of cource be of such use that having wild color transitions do not matter. Combination of infill + surrogate model would be possible, but does require considerable amount of work and there might a issues I've not thought about...