spegelius/filaswitch #34: bug

wbarber69, 2018-01-28:
the latest release gives me this error.
`C:\filaswitch5>filaswitch.bat
Traceback (most recent call last):
  File "C:\filaswitch5\\src\filaswitch.py", line 300, in <module>
    main()
  File "C:\filaswitch5\\src\filaswitch.py", line 271, in main
    gui = GUI()
  File "C:\filaswitch5\\src\filaswitch.py", line 231, in __init__
    self.log = Logger(os.path.join(root_dir, "logs"))
  File "C:\filaswitch5\src\logger.py", line 12, in __init__
    self.filehandler = logging.FileHandler(os.path.join(logdir, "filaswitch.log"))
  File "C:\Users\Walter Barber\AppData\Local\Programs\Python\Python36-32\lib\logging\__init__.py", line 1030, in __init__
    StreamHandler.__init__(self, self._open())
  File "C:\Users\Walter Barber\AppData\Local\Programs\Python\Python36-32\lib\logging\__init__.py", line 1059, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\filaswitch5\\logs\\filaswitch.log'
`

wbarber69, 2018-01-28:
adding the folder and file manually fixed it. but it needs adressing

wbarber69, 2018-01-28:
also spaces in file names are no longer supported

cfelicio, 2018-01-28:
same here, adding the logs folder manually fixed it.

wbarber69, 2018-01-28:
I have confirmed that there is an issue with spaces in file names.  I have reverted to a previous version and it works just fine.

spegelius, 2018-01-28:
Ok logs dir is easy to fix, but more details needed for white space problem: cannot open files that have white space or filaswitch cannot be in folder that has white space? Those both I tested if I recall correctly...

spegelius, 2018-01-28:
Well, tried opening filaswitch extracte a dir that has white space and it starts fine. Sliced a while that's in a dir that has white space, works fine. Using the filaswitch.bat from cmdline seems to cause problems, however

spegelius, 2018-01-28:
Ok, for me the problem as - character in the path, There's a fix attempt in master branch

wbarber69, 2018-01-28:
yes i do have a - in my gcode path but it was cutting my commands off at the spaces and calling anyting between the spaces another command switch even with quotations around the whole thing

wbarber69, 2018-01-28:
bug squashed

closed 2018-01-28 by wbarber69

wbarber69, 2018-01-28:
Swapped out filaswitch.bat for the latest. And the issue seems to have gone away, and s3d post processing happens as normal.  Now just to add in a way to upload it straight to octoprint and we got a finished product here.