spegelius/filaswitch #35: Simplify3D fan is turned on before tower

spegelius, 2018-01-28:
On some cases, like when printing ABS, Simplify3D turns the fan on for some reason and this happens just before tool change -> tower gets cooling -> warping

wbarber69, 2018-01-28:
perhaps its because of differences in nozzle temp on the gcode?

spegelius, 2018-01-28:
No, the models I've been printing all use same temp on the nozzle. It's probably related to the extra cooling on fast layers-feature. I've seen this before, S3D likes to turn the fan on and off somewhat irregularly. Doesn't matter with PLA, but with ABS it's a real problem. I've tuned to fast layer time down to 1 seconds or so, but it still does this occasionally... Compare to Prusa Slic3r, when fan setting is off, it stays off.