spegelius/filaswitch #38: how does the automated bat file work

wbarber69, 2018-01-29:
im currently using the post processing script in s3d to run filaswitch on a completed slice.  But are there other arguments i can pass to make sure the purge tower gets done properly.  if im using the script does it default to 6 lines or does it keep whatever settings i have in the gui?  can i just put 9 somewhere in the script that gets passed so that i can make sure its using enough area to purge?

spegelius, 2018-01-30:
Run filaswitc.bat --help to see all the options. Currently there are couple of options: --lines forl ine count and --position for tower position. Currently lines and position default to 6 and Auto, but they can be changed to read the stored settings.

spegelius, 2018-01-30:
Oh and there's third option, --debug. Normally not needed, but in case of bughunting it prints more info

closed 2018-04-16 by spegelius