spegelius/filaswitch #39: Temperature issues

paukstelis, 2018-02-21:
The most recent version has been working great for me, but I seem to have run into an issue (perhaps this was predictable) when trying to use different temperatures. S3D forces you to assign different Tools for each heater in the Temperature section. This means that all M104 commands are associated with a particular tool. It seems that the Prusa MMU firmware only accepts M104 commands that reference T0. A simple fix would be to have a hardware config setting that would toggle having temp changes all rM104 references to T0 (or no tool at all, which I think also works fine) or have M104 associated with specific tools.

spegelius, 2018-02-22:
Ok, apparently different firmwares handle this differently.
Hw config setting is probably the best option for this, should be easy to implement

spegelius, 2018-02-23:
Committed experimental code to temp_id branch, not tested.

paukstelis, 2018-02-23:
Gcode looks good after just doing a diff. I'll check a print this weekend. Hoping to start looking at regrouping S3D support blocks this weekend as well.

paukstelis, 2018-02-26:
Working nicely for me. I did do a little digging at it seems like the MMU temperature was not working correctly when I was only using 3 extruders. If I was using 4 extruders it worked as expected. Not really sure why. I think this will still be a useful feature moving forward.

spegelius, 2018-02-27:
Ok great, I'll merge this to master.
Did you check the  g-code for 3 extruders print? Was just wondering if S3D had used proper temp id assignments...

paukstelis, 2018-04-19:
Did this get removed from master or did the PRUSAMMU config not get merged from temp_id?

spegelius, 2018-04-21:
Hmm, previous comment missing... anyways, I added this to master, had apparently forgotten to merge it