spegelius/filaswitch #41: Simplify3D supports

paukstelis, 2018-03-04:
S3D supports aren't very smart for multicolor/multimaterial objects. Its a tad annoying to have part of the layer supports printed, followed by a tool change, and then more supports. It would be nicer if all supports were printed first. I just banged this method out quickly to compile all supports for a certain height together and put them in at the first instance of supports at that height. Not sure the logic is most efficient, but it seems to work reasonably well. I don't think this would really change printing time all that much, but it at least feels nicer. I put it in slicer_simplify3d.py and call it right before self.print_summary().

```
def combine_supports(self):
 
        current_height = self.layers[0].height
        #stored layer index and line index
        layer_i = (0,0)
        height_change = False
        layer_supports = []
        delete_index = []
    
        for idx,layer in enumerate(self.layers):
            if current_height != layer.height:
                #on a new height, write our old supports in our stored position
                if len(layer_supports) > 0:
                	thelayer = self.layers[layer_i[0]].lines
                	# do deletion in reverse order first
                	#TODO: this could probably be much better?
                	for la,li in sorted(delete_index, reverse=True):
                		dellayer = self.layers[la].lines
                		del dellayer[li]
                	insert = layer_i[1]+1
                	#slice in compiled supports at the designated position
                	#TODO: probably worth while to add method in layer.py to allow slice insertion
                	thelayer[insert:insert] = layer_supports
                    
                height_change = True
                insert = 0
                layer_supports = []
                delete_index = []
                
            #scan for supports
            m = False
            add = False
            for cmd,comment,l_idx in layer.read_lines():
                if comment:
                	m = self.SUPPORT_RE.match(comment)
                	#adjustment for first layer (idx 0)
                	if (m and height_change) or (m and idx == 0):
                		add = True
                		layer_i = (idx,l_idx)
                		continue
                    
                	elif m:
                		add = True
                		continue
                    
                #compile supports till we hit the next comment only line
                if add and cmd:
                	layer_supports.append((cmd,comment))
                	delete_index.append((idx,l_idx))
                    
                else:
                	add = False
                    
            current_height = layer.height
```

spegelius, 2018-03-12:
Ok I'll test this in include it in if it seems to be working properly.

paukstelis, 2018-03-12:
I'm not sure how useful it really is. I've been using it with no ill effects. What would be more useful to overcome S3D support shortcomings would be to actually reorder all the support lines so they would be spatially connected. Right now it somewhat randomly runs around the bed and puts supports down for different areas. That ends up being a lot more complicated, however.