spegelius/filaswitch #44: Ability to turn off retraction

wbarber69, 2018-04-15:
I would like to give linear advance a try with dual colors.  but with no retraction enabled filaswitch doesnt work.

spegelius, 2018-04-16:
Does fs crash when disabling retraction? It might be that the retraction values aren't in the header when it's disabled and header parsing fails...

About LA: I've been running LA for some time, I even ran the LA calibration g-code to get the correct value. However, I still need some retraction, otherwise it oozes. Also ProStruders don't seem to be best extruders for LA, they sound kinda horrible with LA enabled and K-value 200. I need to drop print acceleration to 500 to get the noise down, with higher acceleration the extruders doesn't seem to be able to keep up.

wbarber69, 2018-04-16:
Linear advance has had some improvements since I see you are using high values.  LA values are now much smaller.  My LA value is 1.6.  Still there may be an instance where I need to turn retraction off and yes fs crashes with no retraction enabled.

spegelius, 2018-04-16:
I haven't upgraded to Marlin 1.1.9 yet so still going with the high K values. I understood that the new LA code in 1.1.9 automatically adjusts the accelerations. I'll need to check that later.

I'll have a look at the retractionless settings, it's a bug and needs to be fixed.

spegelius, 2018-04-18:
Fix committed to master, processes fine on my test model. Not tested in print, though

wbarber69, 2018-04-18:
Good man!  Will give it a go tonight.

closed 2021-02-01 by spegelius