spegelius/filaswitch #46: 0.15 layer height

paukstelis, 2018-04-19:
I have not been to use filaswitch to successful print at 0.15 mm layer heights yet. I've ruled out all hardware issues by printing previously sliced/filaswitched files at 0.2 mm without issue. There doesn't seem to be anything drastically different in the gcode except for the extrusion lengths in the tower. These seem like these might be a little thin and I keep getting jams that seems to then only allow a single filament to print (all other extruders fail; using Prusa MMU). Anyone else having issues?

spegelius, 2018-04-20:
I haven't had problems with layers from 0.1mm to 0.3mm, even when they dynamically change in the print. Also the extrusion amount seems to be correct, the tower is 'filled' properly. The initial, prepurge lines are distributed more loosely by design, I had problems with them overextruding when printing sub-0.2mm layers.
I've had problems with some PLAs jamming when printing slow, so in theory it could be possible that the extrusion rate is just too slow in some cases and causes a jam.
I've been thinking about changing how the sub-0.2mm layer purge is printed to keep the feed rate ~same as with 0.2mm, just print the paths wider. This would alleviate some problems sub-0.2mm layers might have...

paukstelis, 2018-04-20:
Perhaps I spoke too soon. I did another old 0.2 mm and got some jamming, so maybe it is hardware related and being exacerbated by lower layer heights.

spegelius, 2019-06-24:
Been printing a lot with layers ranging from 0.1-0.25 a lot and only problems I've encountered are hw related (jamming which I fixed with filament oiler). Closing this since not fs related.

closed 2019-06-24 by spegelius