spegelius/filaswitch #47: Weird out of place infill on the tower

cfelicio, 2018-04-20:
First of all, I'd like to say THANK YOU so much for the work you have put into this. I just tried the latest version, and I'm quite amazed of how much it has improved over the last few months. The hardware config file, now the flexible layer heights, it's all very exciting, making filaswitch a very valuable tool for those of us that are trying to print with multiple colors, and can't afford a commercial solution.

Using the latest 0.18 release, I noticed an out of place infill on the tower, I'm attaching a screenshot and the gcode, not sure what's going on. I did not notice this on the previous version I was using (0.14). Everything else seems excellent!

[image]

[gcode.zip](https://github.com/spegelius/filaswitch/files/1930372/gcode.zip)

spegelius, 2018-04-20:
Hmm, can't see anything wrong with that picture? The first layer is the brim for the tower, it's size now follows the print settings (or cane be configured in the GUI). The larger brim was added because some people wanted some extra adhesion for the tower and I've had problems with it detaching, too (ABS)

cfelicio, 2018-04-20:
My understanding is that the tower has 2 patterns, one when there is a color change happening (where it does the full purge / color switch), and one pattern for when the color does not change (so it just prints a sparse infill on the tower, like a zig zag).

If you look on the picture above, the tower printed a zig zag pattern, but the part has a full color change happening, so it is out of place.

I tried other parts and it always seems to happen on the same layer, quite odd.

closed 2021-02-01 by spegelius