spegelius/filaswitch #52: S3D sequential printing

sersmax, 2018-07-30:
I wanted to use the feature in S3D for sequential printing. Means that i want to change color not every layer but only every 1-5 Layers. That would a save me a lot of Filament that is used at changing color.
The processing works perfect but stops exactly at the point where no more color changing ist needed. (see pictures)
I wanted to check that myself in the code but it is a bit confusing for me as i don't write Python code that often.
[image]
[image]

spegelius, 2018-08-06:
Send me the gcode and preferably the factory file as well so I can see what happens (and understand what you are doing with the processes...)

sersmax, 2018-08-07:
I can't send you the files because they are too big. I zipped the factory file. 
[MM_Kirby.zip](https://github.com/spegelius/filaswitch/files/2267307/MM_Kirby.zip)
If you want the Gcodes (25MB) too i can try to find a other way to send you those too.
As i tried to say: i want to reduce the filament changes by printing multiple layers before switching the color. This could be done by using the feature "sequential printing"
Thanks for the interest in this.

paukstelis, 2018-08-07:
It is unclear to me what you are trying to do with sequential printing in this case. Doing multiple layers in one color before switching to the next would be nearly impossible. At any point where you had an intersection between two colors your nozzle would hit previously printed layers. It is also not clear to me why you are using the false support extruder for your infill. I don't think you gain anything by doing that.

sersmax, 2018-08-07:
Well i set the sequential printing very low to 0.6 mm. Imo this is no problem for the nozzle to not hit the previous color. Yes the nozzle could hit, but it is hot so it won't slam anywhere.
If i set it to 0,6 i can divine the number of changes by 3. So also the lost PLA for purging to a third .
The lost quality would be worth the reduced changes for me. As i have to use a selfmade Y-Splitter, the risk for a failed color change is pretty high for me too.

spegelius, 2018-09-15:
Finally had a look at the factory file. Turns out the bugfix I did for S3D 4.0.1 was the reason why sequential print was failing. I removed the bugfix since it's only certain version of S3D 4.0.1 and S3D has silently released newer versions of 4.0.1 (which is a dick move if you ask me since there's no way to detect the broken version). So the file should process fine now.

closed 2018-09-26 by spegelius