spegelius/filaswitch #55: Prusa MMU2 Filaswitch

Blanchard3d, 2018-10-15:
@paukstelis Thank you for the start of the MMU2 profile for Filaswitch! 
Attached is a working hwcfg file for the Prusa MMU2 on MK3. 
I am using a .6mm Stainless Steel Nozzle so my temps are a little high (215 C ). 
With this profile I am getting good tips and load/ unloads. 
The MMU2 itself has other problems at times (hopefully firmware updates will fix) but for the most part has been printing good with this file. 
Also there is no need for the starting purges and I have been removing them manually except for the one at the start of the print, maybe it should just be an option to do the others or not in the code. 

On my ending script in S3D I have put "M702 C" to unload the filament otherwise the MMU2 will not unload. 

Prints that were done are also attached :) Oh, and the penguin was taking SlicerPE 10 to 12 hours, I have down to 4 now :) 

[PRUSAMMU2-PLA.txt](https://github.com/spegelius/filaswitch/files/2479201/PRUSAMMU2-PLA.txt)
[image]
[image]

spegelius, 2018-10-15:
Great looking prints, very clean colors, me like.

By the starting purges do you mean the purges that are done before the actual print starts? Those can be disabled in the hwconfig by changing "prerun.prime" option to False

Blanchard3d, 2018-10-15:
Yes, I didn't even notice that. Great, thanks!

paukstelis, 2018-10-15:
Did you also start with the basic factory file that I had included in my fork? @spegelius may want to add and track those files, as most links will direct people here.

Blanchard3d, 2018-10-16:
Yes, I used your fork

spegelius, 2018-10-17:
Yes, the factory file would be nice if you don't mind sharing.

paukstelis, 2018-10-17:
My original hwconfig and factory file are part of a pull request I just
submitted.

closed 2019-10-16 by B3D-P