spegelius/filaswitch #58: Rostock Max V3 Issue with Relative Positioning in Purge Tower

stough, 2018-12-05:
I'm just getting started using my Prometheus System.  I'm using S3D and filaswitch.  The gcode preview looks in S3D looks great, but when I print it from the SD card on my Rostock, I get the attached (see photo).  I have also included the gcode.  Have you heard of any similar issues with a Rostock?  Any thoughts on things that I could fix, change?

Thanks!
Tim

[First-Print_fs.gcode.txt](https://github.com/spegelius/filaswitch/files/2650090/First-Print_fs.gcode.txt)
[image]

spegelius, 2018-12-06:
I have no experience with Rostock. What controller/fw it has? Seems that the controller doesn't like the relative movements...
Not sure how to fix that. On filaswitch side moving to absolute coordinates would require quite a large change and I have no time for that. So check the fw if there's some special things that need to be done to get relative coordinates working.

stough, 2018-12-06:
For now, I'm running a RAMBO aboard with Repetier firmware. I say, for now, because I'm updating to a Duet 2 WiFi 32 bit running RepRap after the holidays. I may put aside my dual extruder experiments until after that. I'll also post to the Prometheus forum and ask if anyone is using a Rostock.

I'll add to this issue with anything that I find. 

Thanks,
Tim

Ginko72, 2023-08-24:
I fixed this issue by changing my mainboard.  Works fine now.  😉

closed 2023-08-24 by Ginko72