spegelius/filaswitch #59: {Question} Guide for install?

Stady234, 2019-02-09:
I am completely lost in how to do the initial setup.  I watched the video https://www.youtube.com/watch?v=q9x75oqI2-4  and it goes from explaining what is going on with Prusa MMU, to "now that we have installed"..

I am using a TronXY X5ST-2E... It was recommended i use this program... but for the life of me.. i am stuck at step 1.  lol

Please!! help me make some sanity of this...

spegelius, 2019-02-10:
Installing the program is simple, just download the latest version from the Release-page. But you'll need to write a HW configuration file for your hotend-extruders setup (hwconfig directory). You can take one of the existing files and start working from there.
Of course you ca ntry one of the existing configs, but I suspect that the X5S extruders cannot handle the ramming speeds. Also I have a X5S-2E myself and it doesn't necessarily need filaswitch as the hotend is a mixing hotend (at least on my machine it is, or was, I've moved to E3Dv6 since) and there's no need to change the filament. Your machine might have different hotend though, seems Tronxy is changing the setups now and then...

Stady234, 2019-02-26:
I have been racking my brain and can not get this to work...
1.I downloaded the "master" and unzipped to c (C:\filaswitch-master)
2. per video i entered the path in the "post" processing -- `C:\filaswitch-master\filaswitch.bat C:\Temp\ X5ST-2E`
as the [output_filepath] shown in the video didnt do anything either.
3. i added the parts to start gcode per instructions...
4. i re-downloaded python 3.5.4 (as i already had 3.7)

I have tried multiple locations to save file, but never get the second "fs" file...

this is the only entry in the "log" file...
2019-02-26 01:52:07,375 - filaswitch - INFO - Detected Simplify3D format
2019-02-26 01:52:07,376 - filaswitch - INFO - HW config: X5ST-2E

spegelius, 2019-02-26:
S3D post processing script works very wonky, I don't use it myself. It might be that you need to make sure that the python executable is in PATH-environment variable.
You can try opening the filaswitch GUI and manually running it, i.e. open the sliced file in the GUI, it should do its thing. There's filaswitch.bat you can run, or filaswitch.sh if running linux

closed 2019-07-24 by Stady234