spegelius/filaswitch #60: Slic3r settings

tenmonkeys, 2019-03-19:
The script can't process Slic3r's default percentage-based settings for extrusion widths and print speeds. The execution stops with "ERROR: Gcode processing failed: Extrusion width cannot be zero
" if at least one of the extrusion widths is not a straight number and "ERROR: Gcode processing failed: could not convert string to float" with printing speeds.

Everything works fine and dandy with if both are defined in mm and mm/s though. But Slic3r PE is pretty smart when it comes to automatically setting extrusion widths, so it'd be a shame to not use that feature.

spegelius, 2019-04-06:
Pushed a experimental fix to new branch. Not tested.
The fix for extrusion widths is simply to use nozzle width when value is percentage; calculating a single value is problematic due to the calculation using layer height and that value can change if using variable layer heights... also when printing the tower the extrusion width doens't play any important role anyways.

spegelius, 2019-06-24:
Released and should be working now. If there's problems still, create new issue

closed 2019-06-24 by spegelius