spegelius/filaswitch #61: prerun.prime false results in division by zero

paukstelis, 2019-04-30:
The default PRUSAMMU2-PLA profile has prerun.prime disabled. However, when this is disabled it throws a division by zero fault.

```
Traceback (most recent call last):
  File "/home/paul/filaswitch/src/filaswitch.py", line 209, in load_file
    self.result_file = pf.process(gcode_file)
  File "/home/paul/filaswitch/src/slicer_simplify3d.py", line 57, in process
    self.add_tool_change_gcode()
  File "/home/paul/filaswitch/src/gcode_file.py", line 320, in add_tool_change_gcode
    for line in self.switch_tower.get_tower_lines(layer, e_pos, self.active_e, new_e):
  File "/home/paul/filaswitch/src/switch_tower.py", line 1035, in get_tower_lines
    gap, purge_multiplier, lines = self._calculate_purge_values(layer_h, new_e)
  File "/home/paul/filaswitch/src/switch_tower.py", line 918, in _calculate_purge_values
    lines = purge_e / (utils.extrusion_feed_rate(self.purge_line_width, layer_h, 1.75) * self.purge_length * 2)
```

spegelius, 2019-05-19:
Generic parser or master?

paukstelis, 2019-05-19:
I believe this was master, but I will have to go back and check.

spegelius, 2019-06-24:
This was fixed in genericparser, master shouldn't have this problem

closed 2019-06-24 by spegelius