spegelius/filaswitch #65: Fan keep on/off in tower

jbaiao, 2019-07-27:
Each time the tower is printed, the fan will turn off and on again when it passes to the object.
Any way to always stay connected?

spegelius, 2019-08-08:
The fan turned off in code for now. I added this functionality because I had some problems with PETG print cooling too much during tool changes and getting bit too brittle.
I could add an option for it in Advanced tab if the feature is causing problems

jbaiao, 2019-08-08:
Yes! That option in advanced tab it's greatest!
By the way, do you know or have a profile for E3d v6 full metal in bowden system for pla or petg?

spegelius, 2019-08-10:
Added new option to advanced-tab, check latest code from master branch.

Prometheus-PTFE_EV6 is a cfg meant for E3D v6 and Prometheus system, so you can try it or use it as a base for your own configuration. Also the PrusaMMU cfg's are for EV6. I don't have any EV6 system yet, I'm building one and trying out what works the best when I get it running.

closed 2021-02-01 by spegelius