spegelius/filaswitch #71: Purge Tower Wrong direction?

TelliMantelli, 2020-04-16:
Purge tower going along y instead of X? Version 0.22
[image]

spegelius, 2020-04-16:
Post your .hwcfg contents

TelliMantelli, 2020-04-16:
Thanks here you go 
`## Hint: [0] means array of values ([0], [1], [2]...)

## Temperature handling
## use_id: should M104 commands have tool id. M104 S210 or M104 S210 T1. Firmware dependent
tool.temperature.use_id:  True

## What command to use for controlling nozzle temperature. M104 and G10 supported
tool.temperature.command: M104

## Nozzle diameter. Used with slicers (Cura) that do not provide settings in gcode files
## Overridden for S3D, Slic3r and KISS
tool.nozzle.diameter: 0.4

## Z-hop when moving to and from the tower. Used with slicers (Cura) that do not provide settings in gcode files.
## Overridden for S3D, Slic3r and KISS
tool.tower.zhop: 0.5

## Machine properties. Used with slicers (Cura) that do not provide settings in gcode files.
# 0 is cartesian, 1 delta
machine.type: 0
machine.x.length: 220
machine.y.length: 220
machine.z.travel.speed: 1000
machine.xy.travel.speed: 1000
machine.origin.offset_x: 0
machine.origin.offset_y: 0
machine.default.speed: 1000

## Extruder motor current values. Adjusts motor current using M907. Values are 
## for load/unload and for normal operation.
## load = motor current for load/unload
## run = motor current for normal operation
## These can be set to 0 if not applicable
motor.current.load: 0
motor.current.run: 0

## These prepurge steps are executed before the actual prepurge. They seem to help with tip formation on some cases
## Disabled by default
#prepurge.initial.retract: 20
#prepurge.initial.retract.speed: 350
#prepurge.initial.pause: 2500

## Prepurge means the fast purge movements that purge old material before tool change
## - temperature = change temperature of the nozzle this amount before doing the prepurge. Negative value lowers the temperature
## - length = length of one sweep
## - extrusion.length = how much material to extrude per sweep
## - speed = how fast to move in one sweep
## - count = how many sweeps
## - gap = distance to next sweep line
prepurge.temperature.change: 0
prepurge.sweep.length: 20
prepurge.sweep.extrusion.length: 4.5
prepurge.sweep.speed: 1000
prepurge.sweep.count: 5
prepurge.sweep.gap: 1
prepurge.sweep.gap.speed: 800

## Prime extruders
## Change next line to True to enable
prerun.prime: true
prerun.prime.length: 49
prerun.prime.extrusion.length: 1.8
prerun.prime.gap: 0.48
prerun.prime.speed: 1000
prerun.prime.xstart: 5.0
prerun.prime.ystart: -3.0
prerun.prime.purge.count: 6

## Rapid retract means pulling the filament out of the nozzle.
## Various stages can be defined.
## - initial retract happens right after prepurge is done
## length and speed are self explanatory
## wipe = do wipe movements during retracts to reduce oozing
## pause = how long to let filament cool in throat (ms)
## long retract = pulling filament fully out of way

rapid.retract.initial[0].length: 10
rapid.retract.initial[0].speed: 800
rapid.retract.wipe: False
rapid.retract.pause: 900
rapid.retract.long[0].length: 12
rapid.retract.long[0].speed: 800

## Various filament cooling movements after major retract. Uncomment to enable. Make sure that there are even amount
## of cooling movements and the total length is 0
## Serves as a wipe
#rapid.retract.cool[0].length: 3
#rapid.retract.cool[0].speed: 1600
#rapid.retract.cool[1].length: -5
#rapid.retract.cool[1].speed: 1600
#rapid.retract.cool[2].length: 5
#rapid.retract.cool[2].speed: 2000
#rapid.retract.cool[3].length: -5
#rapid.retract.cool[3].speed: 2400
#rapid.retract.cool[4].length: 5
#rapid.retract.cool[4].speed: 2400
#rapid.retract.cool[5].length: -3
#rapid.retract.cool[5].speed: 2400

## After tool change, feed new filament
## Use one or more stages (array). Feeding is recommended to do in different speeds:
## - start relatively slow
## - ramp up the speed for the main length of the feed
## - slow down for the last bit for smoother insert

feed[0].length: 10
feed[0].speed: 400
feed[1].length: 25
feed[1].speed: 600
feed[2].length: 27
feed[2].speed: 800

## Do X movement during feeding to prevent blobs
## Change next line to True to enable
feed.trail: True

## prime trail is the first extrusion after filament is almost fully fed to the nozzle.
## Might be useful for clearing old filament if extruding bit more than normally
prime.trail.extrusion.length: 5
prime.trail.speed: 900

## Tower coasting length. Use to even out the extrusion after tower if there's over-extrusion seen.
post.tower.coast: 0.2`

spegelius, 2020-04-16:
Change prepurge.sweep.length: 20 to 50 or something

TelliMantelli, 2020-04-17:
Excellent! Thanks man great stuff appreciate all the work!

closed 2020-04-17 by TelliMantelli