spegelius/filaswitch #72: No Zhop even if enabled..

TelliMantelli, 2020-04-18:
I cannot get zhop to work between print and tower. I have enable in my config. Do I also enable in S3D?
Thanks!

TelliMantelli, 2020-04-19:
Here's a link to my config and gcode. https://drive.google.com/open?id=1ucZ8YRqnAdrLgz80fHe8iYcGjfakfqmT

spegelius, 2020-04-19:
Currently filaswitch follows the slicer settings for z-hop (except for Cura). So if you don't have z-hop enabled in S3D, it won't do it with tower.

TelliMantelli, 2020-04-19:
Ahhh ok. So as of right now there no way to have zhop between model and tower only..not between each layers also with S3D?

spegelius, 2020-04-19:
Yes, but that could be changed so that hwcfg tower z-hop value is used always.

TelliMantelli, 2020-04-19:
That be great to have it work that way! Be willing to test.

closed 2020-04-19 by TelliMantelli