spegelius/filaswitch #73: Layer shift from build up on Tower 2 PLaces.

TelliMantelli, 2020-04-19:
On my pruge tower you can see from added photos that on the purge tower Im getting build ups in 2 spots which cause a layer shift. What can I do to reduce this behavoir? Thanks really appreciate all the help lately!
[image]
[image]

spegelius, 2020-04-20:
Yuo probably have the new filament feed too long so it builds up too much material. So in hwcfg. make sure the total rapid.retract length is more that the total feed length. Looking at the values you posted, you are feeding quite a lot more that you are retracting, almost 3 times which will cause filament buildup

TelliMantelli, 2020-04-21:
Ok awesome let me make those changes! Thanks man again really appreciate the help. If I get this worked out my A10m hwcfg should work with all the M series as the only difference is bed size.

closed 2020-04-23 by TelliMantelli