spegelius/filaswitch #75: TypeError: float() argument must be a string or a number, not 'NoneType'

momo9700, 2021-03-24:
I'm currently using the last version of S3D on MacOS and it says this error.

spegelius, 2021-03-25:
S3D version 5? I don't have that. Can you send me the gcode file for debugging?

momo9700, 2021-03-25:
the version of S3D is the 4.1.2.
It gives me this error whit any gcode i prepare whit dual extrusion wizard in the slicer

spegelius, 2021-03-29:
Sorry about the late reply, but I need a gcode file that fails so I can replicate the error. Also if you are using a custom hwcfg file, please send that also

momo9700, 2021-03-29:
no worries. here is the files.
[files.zip](https://github.com/spegelius/filaswitch/files/6220764/files.zip)

spegelius, 2021-04-04:
Ok it seems the line 'prepruge.temperature.change: -10' has a typo (prepruge). I'll update filaswitch with better error information.

momo9700, 2021-04-06:
thanks very much

closed 2021-04-11 by spegelius